xyzroe/XZG #169: CC2652P7 Included Thread Firmware Issue

lty1993, 2025-07-09:
**Describe the bug**
Included CC2652P7 Thread RCP firmware will stop response UART messages after some device connecting to OTBR. 

**Device information**
- Screenshot of Zigbee and Device tab from root page of device

[image]

[image]

- JSON object returned on [http://xzg.local/api?action=1&param=all](http://xzg.local/api?action=1&param=all) 
 *You **should** remove your credentials first*

```
{
  "network-config": {
    "wifiEnbl": false,
    "wifiSsid": "",
    "wifiPass": "",
    "wifiDhcp": true,
    "wifiIp": "::",
    "wifiMask": "255.255.255.0",
    "wifiGate": "::",
    "wifiDns1": "1.1.1.1",
    "wifiDns2": "8.8.8.8",
    "wifiPwr": 78,
    "wifiMode": 1,
    "ethEnbl": true,
    "ethDhcp": true,
    "ethIp": "::",
    "ethMask": "255.255.255.0",
    "ethGate": "::",
    "ethDns1": "1.1.1.1",
    "ethDns2": "8.8.8.8"
  },
  "vpn-config": {
    "wgEnable": false,
    "wgLocalIP": "::",
    "wgLocalSubnet": "255.255.255.0",
    "wgLocalPort": 33333,
    "wgLocalGateway": "0.0.0.0",
    "wgLocalPrivKey": "",
    "wgEndAddr": "",
    "wgEndPubKey": "",
    "wgEndPort": 0,
    "wgAllowedIP": "0.0.0.0",
    "wgAllowedMask": "0.0.0.0",
    "wgMakeDefault": true,
    "wgPreSharedKey": "",
    "hnEnable": false,
    "hnJoinCode": "",
    "hnHostName": "XZG-AD39",
    "hnDashUrl": "default"
  },
  "mqtt-config": {
    "enable": false,
    "server": "",
    "port": 1883,
    "user": "",
    "pass": "",
    "topic": "XZG-AD39",
    "updateInt": 60,
    "discovery": true,
    "reconnectInt": 15
  },
  "system-config": {
    "disableWeb": false,
    "webAuth": true,
    "webUser": "654321",
    "webPass": "123456",
[... 61 lines cut ...]
[11273030] -> 7e 81 05 84 26 04 e8 14 56 7e
[11273034] <- 7e 81 08 84 26 04 e8 60 2a 7e
[11273043] -> 7e 82 04 85 26 72 e3 a5 4f 25 f9 c1 02 ae c8 7e
[11273046] <- 7e 82 07 85 26 72 e3 a5 4f 25 f9 c1 02 1d 36 7e
[11273138] <- 7e 80 06 71 22 00 6b dc 84 65 79 8e 25 0e 1a 7c 7d 5d 68 b2 02 c1 f9 25 4f a5 e3 72 0d 8c 00 00 00 01 04 dd 6b 9f b9 c4 3b c4 80 10 00 0a 00 0f 55 55 ee 25 03 00 00 00 00 01 00 00 05 00 00 00 00 00 00 f2 a5 7e
[11273146] -> 7e 83 03 71 7d 5e 00 79 dc 6b 65 79 02 c1 f9 25 4f a5 e3 72 8e 25 0e 1a 7c 7d 5d 68 b2 0d 33 f0 4d 4c 01 c0 b2 eb 45 7f 33 f0 4d 4c 4d 4c 2a a2 00 15 8d 5a 00 00 00 00 00 00 01 80 7f 1e e0 1d 3e 6b ab 94 6b 87 e8 e7 70 32 03 bf c6 d5 80 b2 6f 4b 28 9d e7 63 e6 76 26 f6 43 e0 78 cf bc 90 ac 2b ea cb 29 8b 5e 05 ca a6 82 7a 38 6c 63 17 9a 91 de fb f2 54 29 d0 cb 31 10 dc 42 4d 68 e9 8b 00 00 00 00 00 0f 04 00 01 00 00 00 00 00 00 00 00 00 00 00 0f 7f 55 37 7e
[11273157] <- 7e 83 06 00 7d 31 00 01 01 73 47 00 00 fd ab 7e
[11273198] <- 7e 80 06 71 22 00 6b dc 85 65 79 8e 25 0e 1a 7c 7d 5d 68 b2 02 c1 f9 25 4f a5 e3 72 0d 8d 00 00 00 01 04 1e 36 d5 1a c5 39 c5 80 10 00 0a 00 0f 59 00 dd 26 03 00 00 00 00 01 00 00 05 00 00 00 00 00 00 06 50 7e
[11273207] -> 7e 84 03 71 7d 5e 00 79 dc 6b 65 79 02 c1 f9 25 4f a5 e3 72 8e 25 0e 1a 7c 7d 5d 68 b2 0d 73 47 00 00 01 c0 b2 eb 45 7f 33 f0 4d 4c 4d 4c 2a a2 00 15 8d 5a 00 00 00 00 00 00 01 80 7f 1e e0 1d 3e 6b ab 94 6b 87 e8 e7 70 32 03 bf c6 d5 80 b2 6f 4b 28 9d e7 63 e6 76 26 f6 43 e0 78 cf bc 90 ac 2b ea cb 29 8b 5e 05 ca a6 82 7a 38 6c 63 17 9a 91 de fb f2 54 29 d0 cb 31 10 dc 42 4d 68 e9 8b 00 00 00 00 00 0f 04 00 01 01 01 00 00 00 00 00 00 00 00 00 0f 7f 33 03 7e
[11273219] <- 7e 84 06 00 7d 31 00 01 01 73 47 00 00 1b 0b 7e
[11273231] <- 7e 80 06 71 22 00 6b dc 86 65 79 8e 25 0e 1a 7c 7d 5d 68 b2 02 c1 f9 25 4f a5 e3 72 0d 8e 00 00 00 01 04 ad 10 1b d1 c5 3b c5 80 10 00 0a 00 0f 59 8b 5c 27 03 00 00 00 00 01 00 00 05 00 00 00 00 00 00 7d 5e 14 7e
[11273239] -> 7e 85 03 71 7d 5e 00 79 dc 6b 65 79 02 c1 f9 25 4f a5 e3 72 8e 25 0e 1a 7c 7d 5d 68 b2 0d 73 47 00 00 01 c0 b2 eb 45 7f 33 f0 4d 4c 4d 4c 2a a2 00 15 8d 5a 00 00 00 00 00 00 01 80 7f 1e e0 1d 3e 6b ab 94 6b 87 e8 e7 70 32 03 bf c6 d5 80 b2 6f 4b 28 9d e7 63 e6 76 26 f6 43 e0 78 cf bc 90 ac 2b ea cb 29 8b 5e 05 ca a6 82 7a 38 6c 63 17 9a 91 de fb f2 54 29 d0 cb 31 10 dc 42 4d 68 e9 8b 00 00 00 00 00 0f 04 00 01 01 01 00 00 00 00 00 00 00 00 00 0f 7f 2c b2 7e
[11273250] <- 7e 85 06 00 7d 31 00 01 01 73 47 00 00 8a 5e 7e
[11273291] <- 7e 80 06 71 22 00 6b dc 87 65 79 8e 25 0e 1a 7c 7d 5d 68 b2 02 c1 f9 25 4f a5 e3 72 0d 8f 00 00 00 01 04 94 d4 e4 af c4 3c c4 80 10 00 0a 00 0f 55 3c 4a 28 03 00 00 00 00 01 00 00 05 00 00 00 00 00 00 7d 5d 9b 7e
[11273300] -> 7e 86 03 71 7d 5e 00 79 dc 6b 65 79 02 c1 f9 25 4f a5 e3 72 8e 25 0e 1a 7c 7d 5d 68 b2 0d 73 47 00 00 01 c0 b2 eb 45 7f 33 f0 4d 4c 4d 4c 2a a2 00 15 8d 5a 00 00 00 00 00 00 01 80 7f 1e e0 1d 3e 6b ab 94 6b 87 e8 e7 70 32 03 bf c6 d5 80 b2 6f 4b 28 9d e7 63 e6 76 26 f6 43 e0 78 cf bc 90 ac 2b ea cb 29 8b 5e 05 ca a6 82 7a 38 6c 63 17 9a 91 de fb f2 54 29 d0 cb 31 10 dc 42 4d 68 e9 8b 00 00 00 00 00 0f 04 00 01 01 01 00 00 00 00 00 00 00 00 00 0f 7f 1c 69 7e
[11273312] <- 7e 86 06 00 7d 31 00 01 01 73 47 00 00 39 a0 7e
[11273315] -> 7e 87 05 85 26 02 c1 f9 25 4f a5 e3 72 dd 5d 7e
[11275317] -> 7e 81 03 20 01 d5 10 7e
[11277320] -> 7e 81 03 20 01 d5 10 7e
[11282949] | Connected client 1 from 172.18.6.6
[11283590] -> 7e 80 01 02 ea f0 7e
[11285592] -> 7e 81 02 01 c5 b2 7e
[11287596] -> 7e 81 02 01 c5 b2 7e
[11289597] -> 7e 81 02 01 c5 b2 7e
```

xyzroe, 2025-07-09:
Please write an issue [here](https://github.com/Koenkk/OpenThread-TexasInstruments-firmware/issues)

xyzroe, 2026-11-15:
Closing: the Thread RCP firmware itself comes from the upstream TI/OpenThread builds, XZG only flashes it. Please report RCP stability problems upstream.

closed 2026-11-15 by xyzroe